moveit/moveit_ikfast #49: ValueError when creating plugin under Ubuntu 14.04 and ROS Indigo

pirobot, 2015-05-08:
My setup is Ubuntu 14.04 and ROS Indigo Debian install.  I am using version 0.9 of OpenRave, source install since Debian packages for Ubuntu 14.04 are not yet available.

After building the .cpp file using OpenRave's ipfast.py utility, I get the following error when trying to build the plugin:

$ rosrun moveit_ikfast create_ikfast_moveit_plugin.py pi_robot right_arm pi_robot_ikfast_right_arm_plugin `pwd`/output_ikfast61.cpp
Warning: The default search has changed from OPTIMIZE_FREE_JOINT to now OPTIMIZE_MAX_JOINT!

IKFast Plugin Generator
Loading robot from 'pi_robot_moveit_config' package ... 
Creating plugin in 'pi_robot_ikfast_right_arm_plugin' package ... 
  found 2 planning groups: right_arm, right_gripper
  found group 'right_arm'
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/moveit_ikfast/create_ikfast_moveit_plugin.py", line 178, in <module>
    solver_version = int(line_search.group(1))
ValueError: invalid literal for int() with base 10: '0x10000048'

My output_ikfast.cpp file can be found [here](https://www.dropbox.com/s/in4mvh65n9ry5t9/output_ikfast61.cpp?dl=0).

gavanderhoorn, 2015-05-08:
I'm pretty sure this has already been fixed when we merged #48. It just hasn't been released yet.

You don't mention it, but did you install a released version of `moveit_ikfast` (debs)? If so, please try a from-source build and see whether that allows you to continue.

pirobot, 2015-05-08:
Thanks @gavanderhoorn -- building from source works.  And yes, I should have mentioned I was using the Indigo deb version.

closed 2015-05-08 by pirobot

mikeferguson, 2015-05-08:
Should we push a new indigo release then?

gavanderhoorn, 2015-05-08:
I guess we could. Would include #48 and some other [minor updates](https://github.com/ros-planning/moveit_ikfast/compare/3.1.0...indigo-devel).

chamberger, 2015-05-26:
Hi @pirobot, can you provide me the detailed steps on installing ikfast and openrave on ubuntu14.04 with ROS indigo? Thanks in advanced for your reply!

gavanderhoorn, 2015-05-26:
@chamberger: I hate to sound like a broken record, but could you please ask this question on [ROS Answers](http://answers.ros.org/questions/)? Not only would that help others, it would also not hijack this issue with something that is at best tangential.

Thanks.