moveit/moveit_ikfast #54: catkin_make plugin fails with ‘kinematics::KinematicsResult’ has not been declared

pirobot, 2015-06-28:
I'm trying to get my IKFast plugin to build under Ubuntu 14.04 and ROS Indigo debian install.  I am using the source install of moveit_ikfast because of a bug in the debian version that has been fixed in source but not yet pushed to the debs.

As for OpenRave, I am using a source install from the master branch which compiles fine and generates my .cpp file.

After creating my empty plugin package, running catkin_make yields the following error which causes the plugin not to be built:

/home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:178:42: error: ‘kinematics::KinematicsResult’ has not been declared

The rest of the output is as follows:

                                  kinematics::KinematicsResult& result,
                                              ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:302:41: error: ‘kinematics::DiscretizationMethod’ has not been declared
       bool sampleRedundantJoint(kinematics::DiscretizationMethod method, std::vector<double>& sampled_joint_vals) const;
                                             ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp: In constructor ‘ikfast_kinematics_plugin::IKFastKinematicsPlugin::IKFastKinematicsPlugin()’:
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:139:5: error: ‘supported_methods_’ was not declared in this scope
         supported_methods_.push_back(kinematics::DiscretizationMethods::NO_DISCRETIZATION);
         ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:139:46: error: ‘kinematics::DiscretizationMethods’ has not been declared
         supported_methods_.push_back(kinematics::DiscretizationMethods::NO_DISCRETIZATION);
                                                  ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:140:46: error: ‘kinematics::DiscretizationMethods’ has not been declared
         supported_methods_.push_back(kinematics::DiscretizationMethods::ALL_DISCRETIZED);
                                                  ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:141:46: error: ‘kinematics::DiscretizationMethods’ has not been declared
         supported_methods_.push_back(kinematics::DiscretizationMethods::ALL_RANDOM_SAMPLED);
                                                  ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp: In member function ‘void ikfast_kinematics_plugin::IKFastKinematicsPlugin::setSearchDiscretization(const std::map<int, double>&)’:
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:456:3: error: ‘redundant_joint_discretization_’ was not declared in this scope
       redundant_joint_discretization_.clear();
       ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp: At global scope:
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1058:56: error: ‘kinematics::KinematicsResult’ has not been declared
                                                kinematics::KinematicsResult& result,
                                                            ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp: In member function ‘bool ikfast_kinematics_plugin::IKFastKinematicsPlugin::getPositionIK(const std::vector<geometry_msgs::Pose_<std::allocator<void> > >&, const std::vector<double>&, std::vector<std::vector<double> >&, int&, const kinematics::KinematicsQueryOptions&) const’:
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1066:12: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
         result.kinematic_error = kinematics::KinematicErrors::SOLVER_NOT_ACTIVE;
                ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1066:42: error: ‘kinematics::KinematicErrors’ has not been declared
         result.kinematic_error = kinematics::KinematicErrors::SOLVER_NOT_ACTIVE;
                                              ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1073:12: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
         result.kinematic_error = kinematics::KinematicErrors::EMPTY_TIP_POSES;
                ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1073:42: error: ‘kinematics::KinematicErrors’ has not been declared
         result.kinematic_error = kinematics::KinematicErrors::EMPTY_TIP_POSES;
                                              ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1080:12: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
         result.kinematic_error = kinematics::KinematicErrors::MULTIPLE_TIPS_NOT_SUPPORTED;
                ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1080:42: error: ‘kinematics::KinematicErrors’ has not been declared
         result.kinematic_error = kinematics::KinematicErrors::MULTIPLE_TIPS_NOT_SUPPORTED;
                                              ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1105:16: error: ‘const struct kinematics::KinematicsQueryOptions’ has no member named ‘discretization_method’
         if(options.discretization_method == kinematics::DiscretizationMethods::NO_DISCRETIZATION &&
                    ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1105:53: error: ‘kinematics::DiscretizationMethods’ has not been declared
         if(options.discretization_method == kinematics::DiscretizationMethods::NO_DISCRETIZATION &&
                                                         ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1114:16: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
             result.kinematic_error = kinematics::KinematicErrors::IK_SEED_OUTSIDE_LIMITS;
                    ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1114:46: error: ‘kinematics::KinematicErrors’ has not been declared
             result.kinematic_error = kinematics::KinematicErrors::IK_SEED_OUTSIDE_LIMITS;
                                                  ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1123:38: error: ‘const struct kinematics::KinematicsQueryOptions’ has no member named ‘discretization_method’
         if(!sampleRedundantJoint(options.discretization_method,sampled_joint_vals))
                                          ^
[... 19 lines cut ...]
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1204:63: error: ‘bool ikfast_kinematics_plugin::IKFastKinematicsPlugin::sampleRedundantJoint’ is not a static member of ‘class ikfast_kinematics_plugin::IKFastKinematicsPlugin’
     bool IKFastKinematicsPlugin::sampleRedundantJoint(kinematics::DiscretizationMethod method, std::vector<double>& sampled_joint_vals) const
                                                                   ^
    /home/patrick/catkin_ws/src/pi_robot_ikfast_right_arm_plugin/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp:1204:51: error: ‘DiscretizationMethod’ is not a member of ‘kinematics’
     bool IKFastKinematicsPlugin::sampleRedundantJoint(kinematics::DiscretizationMethod method, std::vector<double>& sampled_joint_vals) const
                                                       ^
[... cut ...]
     PLUGINLIB_EXPORT_CLASS(ikfast_kinematics_plugin::IKFastKinematicsPlugin, kinematics::KinematicsBase);
                                                                                                         ^
    make[2]: *** [pi_robot_ikfast_right_arm_plugin/CMakeFiles/pi_robot_right_arm_moveit_ikfast_plugin.dir/src/pi_robot_right_arm_ikfast_moveit_plugin.cpp.o] Error 1
    make[1]: *** [pi_robot_ikfast_right_arm_plugin/CMakeFiles/pi_robot_right_arm_moveit_ikfast_plugin.dir/all] Error 2
    make: *** [all] Error 2
    Invoking "make -j4 -l4" failed

gavanderhoorn, 2015-06-28:
Could you add which specific commit you are at? Some changes were recently merged that may affect this, so it's important to know which particular version you are using.

jrgnicho, 2015-06-28:
The PR #51 introduced some significant changes that are related to recent api changes in the KinematicsBase class in moveit_core.  However,  those changes were merged into jade-devel only and so the indigo-devel branch should have not been affected.

pirobot, 2015-06-28:
@gavanderhoorn, this is my commit info:

commit b8edeb68fc5c41bc4410f8ae47c3b1365495fdfb
Merge: 8f650b1 7cdd86f
Author: Dave Coleman <davetcoleman@gmail.com>
Date:   Wed Jun 24 23:39:56 2015 -0700

    Merge pull request #51 from jrgnicho/kinematic_base_changes
    
    KinematicsBase changes

jrgnicho, 2015-06-28:
The changes in that commit should be part of jade-devel only.  Is there anyway that you could use the indigo-devel branch instead?

pirobot, 2015-06-28:
Ooops--my bad.  Did not notice that the default distribution is now jade-devel.  Sorry about that.  Works fine now that I am using indigo-devel.  Closing the ticket.

closed 2015-06-28 by pirobot

gavanderhoorn, 2015-06-29:
@pirobot & @jrgnicho: that was exactly why I asked. Glad that it worked for you after all @pirobot.