moveit/moveit_ikfast #55: ‘kinematics::KinematicsResult’ has not been declared

Ristridin, 2015-08-14:
Hi!
All in all similar to #54 however I'm using ROS Jade on trusty. So the jade branch should be the right one for me.
Any suggestions?

Output as follows:
```
[...]
[100%] Built target moveit_setup_assistant
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:178:42: error: ‘kinematics::KinematicsResult’ has not been declared
                              kinematics::KinematicsResult& result,
                                          ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:302:41: error: ‘kinematics::DiscretizationMethod’ has not been declared
   bool sampleRedundantJoint(kinematics::DiscretizationMethod method, std::vector<double>& sampled_joint_vals) const;
                                         ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp: In constructor ‘ikfast_kinematics_plugin::IKFastKinematicsPlugin::IKFastKinematicsPlugin()’:
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:139:5: error: ‘supported_methods_’ was not declared in this scope
     supported_methods_.push_back(kinematics::DiscretizationMethods::NO_DISCRETIZATION);
     ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:139:46: error: ‘kinematics::DiscretizationMethods’ has not been declared
     supported_methods_.push_back(kinematics::DiscretizationMethods::NO_DISCRETIZATION);
                                              ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:140:46: error: ‘kinematics::DiscretizationMethods’ has not been declared
     supported_methods_.push_back(kinematics::DiscretizationMethods::ALL_DISCRETIZED);
                                              ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:141:46: error: ‘kinematics::DiscretizationMethods’ has not been declared
     supported_methods_.push_back(kinematics::DiscretizationMethods::ALL_RANDOM_SAMPLED);
                                              ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp: In member function ‘void ikfast_kinematics_plugin::IKFastKinematicsPlugin::setSearchDiscretization(const std::map<int, double>&)’:
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:456:3: error: ‘redundant_joint_discretization_’ was not declared in this scope
   redundant_joint_discretization_.clear();
   ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp: At global scope:
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1058:56: error: ‘kinematics::KinematicsResult’ has not been declared
                                            kinematics::KinematicsResult& result,
                                                        ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp: In member function ‘bool ikfast_kinematics_plugin::IKFastKinematicsPlugin::getPositionIK(const std::vector<geometry_msgs::Pose_<std::allocator<void> > >&, const std::vector<double>&, std::vector<std::vector<double> >&, int&, const kinematics::KinematicsQueryOptions&) const’:
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1066:12: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
     result.kinematic_error = kinematics::KinematicErrors::SOLVER_NOT_ACTIVE;
            ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1066:42: error: ‘kinematics::KinematicErrors’ has not been declared
     result.kinematic_error = kinematics::KinematicErrors::SOLVER_NOT_ACTIVE;
                                          ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1073:12: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
     result.kinematic_error = kinematics::KinematicErrors::EMPTY_TIP_POSES;
            ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1073:42: error: ‘kinematics::KinematicErrors’ has not been declared
     result.kinematic_error = kinematics::KinematicErrors::EMPTY_TIP_POSES;
                                          ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1080:12: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
     result.kinematic_error = kinematics::KinematicErrors::MULTIPLE_TIPS_NOT_SUPPORTED;
            ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1080:42: error: ‘kinematics::KinematicErrors’ has not been declared
     result.kinematic_error = kinematics::KinematicErrors::MULTIPLE_TIPS_NOT_SUPPORTED;
                                          ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1105:16: error: ‘const struct kinematics::KinematicsQueryOptions’ has no member named ‘discretization_method’
     if(options.discretization_method == kinematics::DiscretizationMethods::NO_DISCRETIZATION &&
                ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1105:53: error: ‘kinematics::DiscretizationMethods’ has not been declared
     if(options.discretization_method == kinematics::DiscretizationMethods::NO_DISCRETIZATION &&
                                                     ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1114:16: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
         result.kinematic_error = kinematics::KinematicErrors::IK_SEED_OUTSIDE_LIMITS;
                ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1114:46: error: ‘kinematics::KinematicErrors’ has not been declared
         result.kinematic_error = kinematics::KinematicErrors::IK_SEED_OUTSIDE_LIMITS;
                                              ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1123:38: error: ‘const struct kinematics::KinematicsQueryOptions’ has no member named ‘discretization_method’
     if(!sampleRedundantJoint(options.discretization_method,sampled_joint_vals))
                                      ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1125:14: error: request for member ‘kinematic_error’ in ‘result’, which is of non-class type ‘int’
[... 19 lines cut ...]
 bool IKFastKinematicsPlugin::sampleRedundantJoint(kinematics::DiscretizationMethod method, std::vector<double>& sampled_joint_vals) const
                                                               ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1204:51: error: ‘DiscretizationMethod’ is not a member of ‘kinematics’
 bool IKFastKinematicsPlugin::sampleRedundantJoint(kinematics::DiscretizationMethod method, std::vector<double>& sampled_joint_vals) const
                                                   ^
/media/jph/linuxOnlyData/spacebot/src/crustcrawler_arm/moveit_ik_plugin/src/arm_manipulator_ikfast_moveit_plugin.cpp:1204:111: error: expected primary-expression before ‘&’ token
 bool IKFastKinematicsPlugin::sampleRedundantJoint(kinematics::DiscretizationMethod method, std::vector<double>& sampled_joint_vals) const
                                                                                                               ^
[... cut ...]
 PLUGINLIB_EXPORT_CLASS(ikfast_kinematics_plugin::IKFastKinematicsPlugin, kinematics::KinematicsBase);
                                                                                                     ^
make[2]: *** [crustcrawler_arm/moveit_ik_plugin/CMakeFiles/arm_manipulator_moveit_ikfast_plugin.dir/src/arm_manipulator_ikfast_moveit_plugin.cpp.o] Error 1
make[1]: *** [crustcrawler_arm/moveit_ik_plugin/CMakeFiles/arm_manipulator_moveit_ikfast_plugin.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

jrgnicho, 2015-08-14:
@Ristridin , it appears that the changes to the [KinematicsBase interface ](https://github.com/ros-planning/moveit_core/pull/248) haven't made it to the jade debian release yet.  I suggest you use the indigo-devel branch for now until those changes make it to the debian.

Ristridin, 2015-08-14:
@jrgnicho thanks a lot, it worked now :)

closed 2015-08-14 by Ristridin