moveit/moveit_ikfast #56: IKFast release into indigo

sachinchitta, 2016-01-30:
I see differences between indigo and jade for this package. Was the intention to sync them before doing a release? @jrgnicho

jrgnicho, 2016-02-01:
@sachinchitta I was not aware that these changes were made.  I can sync those changes into jade-devel before the release.

gavanderhoorn, 2016-02-02:
What changes are we talking about here exactly? Cause looking at the diff between [indigo-devel and jade-devel](https://github.com/ros-planning/moveit_ikfast/compare/indigo-devel...jade-devel), there are only commits from #51 (and two minor ones from me) that are not supposed to go into `indigo-devel` afaict.

jrgnicho, 2016-02-02:
@gavanderhoorn I was referring to the minor documentation changes, but you are right there are also changes from #51 .  If I remember correctly the changes in #51 should've been merge to indigo-devel (see #50) as as part of an change in the moveit_core/KinematicsBase that involved multiple repos.

gavanderhoorn, 2016-02-02:
Looking at the comment history of #51 (and also ros-planning/moveit_core#248), it would seem as if we collectively decided *not* to merge #50 (into `indigo-devel`), which was why #51 was opened. This is also how I remember it.

We can probably merge this into `indigo-devel` as well, but afaict, all other related changes have only gone into `jade-devel` (in `ros-planning/moveit_core` fi).

jrgnicho, 2016-02-02:
From briefly looking at the commit history in the indigo-devel branches, I can see that the related changes in [moveit_ros 10/03/2015](https://github.com/ros-planning/moveit_ros/commits/indigo-devel/robot_interaction/include/moveit/robot_interaction) and int [moveit_core 10/08/2015](https://github.com/ros-planning/moveit_core/commits/indigo-devel/kinematics_base) were merged.  I think we should merge here in order to be consistent with the changes made in the related repositories.

jrgnicho, 2016-02-23:
@gavanderhoorn , I can go ahead and create a new PR to merge the changes in #50 into indigo-devel if that is OK with you

gavanderhoorn, 2016-03-03:
I feel it's strange to introduce these changes into an LTS release, but seeing as the related PRs have been merged into `indigo-devel` branches in other repositories, we'll have to do that in `moveit_ikfast` as well.

@jrgnicho: if you can provide a PR for this?

jrgnicho, 2016-03-04:
@gavanderhoorn yes, I can do that.

davetcoleman, 2016-06-27:
This issue can be closed because its related PR, #57, was closed

closed 2016-06-27 by davetcoleman