moveit/moveit_ikfast #58: Use of StringStream in new "GetAllIK" Function

Jmeyer1292, 2016-04-21:
So I've been building the software on the Descartes side necessary to interface with the changes in #57 and it's Jade equivalent.

One of the debug statements is built from a `std::stringstream`, and in an optimized build of a workspace, running on a representative problem, this stringstream is eating more than 60% of the runtime involved with the IK calls.
[image]

All of this for:
```c++
ROS_DEBUG_NAMED("ikfast","Sol %d: %s", s, ss.str().c_str());
```

davetcoleman, 2016-04-23:
woa, good find!!

gavanderhoorn, 2016-04-26:
Keeping this open until fe8e790 is cherry-picked into `indigo-devel` as well.