moveit/moveit_ikfast #63: Release into ROS Jade

130s, 2016-06-26:
It's required in https://github.com/ros-planning/moveit_ros/issues/692.

If the branches are ready to be released, I can make releases. Right now I see https://github.com/ros-planning/moveit_ikfast/issues/56 that seems better included for the new release?
Let me know.

Thanks!

davetcoleman, 2016-06-27:
#56 should have been closed

130s, 2016-06-28:
Ok, now running pre-release test locally (for Indigo. I noticed that the last release for Indigo was in 2014 and quite a few updates since then).

Note that I'll increment minor version of Jade to 3.2 since it's on separate branch.

130s, 2016-06-28:
Done
I https://github.com/ros/rosdistro/pull/12014
J https://github.com/ros/rosdistro/pull/12015

closed 2016-06-28 by 130s

davetcoleman, 2016-06-28:
Thank you Isaac!