moveit/moveit_ikfast #68: IKP_TranslationXAxisAngle4D and IKP_TranslationYAxisAngle4D not implemented

robosamir, 2016-07-27:
I'd like to try using these IK types. If implemented, I can try testing them to see if they work.

v4hn, 2016-07-27:
I'm not sure I understand the issue.
Do you want to implement it and provide a patch?
Or do you want *someone* to implement it for you?
There already is a `ROS_ERROR` for these methods not being implemented in the code.
I see no reason for an additional issue unless there is some proposed code for it...

robosamir, 2016-07-27:
I wouldn't mind implementing it but I don't know the logic. Some guidance would be appreciated.

jrgnicho, 2016-07-27:
Are you implementing this for a specific robot?

robosamir, 2016-07-27:
I have a 5DOF arm for which TranslationDirection5D doesn't work but these IK types do.

jrgnicho, 2016-08-01:
This repo only provides tools for wrapping an ik closed form solution into a moveit KinematicsBase subclass.  In order to generate the ik solution you can use the openrave tools as described [here](http://docs.ros.org/hydro/api/moveit_ikfast/html/doc/ikfast_tutorial.html).  As far as I know openrave is very flexible and can solve the ik for arms with varying kinematic configurations.  However, I've only used it with 6DOF arms with with a spherical wrist configuration but I'd advice to try and see if it can generate one for your robot.

robosamir, 2016-08-02:
Yes, I understand that. Anyways, I implemented the logic for IKP_TranslationXAxisAngle4D/IKP_TranslationYAxisAngle4D/IKP_TranslationZAxisAngle4D in the template file. I can make a PR if you are interested.
This issue can be closed.

closed 2016-08-02 by robosamir

v4hn, 2016-08-02:
@robosamir do that please.
Evidence that your implementation actually works (e.g. a moveit config for such a robot) would be highly appreciated!